ethersex/ethersex #382: DHCP static configuration fallback needs EEPROM

tu500, 2015-01-10:
`dhcp_set_static` uses uninitialized memory to set the IPv4 configuration if `EEPROM_SUPPORT` is not active and `DHCP_SUPPORT` does not force `EEPROM_SUPPORT`. Therefore it causes a compiler warning in this case.
`dhcp_net_main` calls `dhcp_set_static` as a fallback on repeated failure of the DHCP request.
However I'm not sure what the right fix would be. I don't see requiring `EEPROM_SUPPORT` and using the stored data there as a fallback to be useful in most cases. I suggest a config option for which data to use as fallback (static config, eeprom, disable ethernet). Opinions?
Related question: Is there a way to set the eeprom data from within make menuconfig to be flashed with the firmware?

eku, 2015-01-10:
Yes, the code is realy a mess! Have a look at the [bootp code] (https://github.com/ethersex/ethersex/blob/master/protocols/bootp/bootp.c#L156) how it could be rewritten to work with and without EEPROM_SUPPORT.

eku, 2015-01-28:
Will you provide a patch?

tu500, 2015-02-10:
If I have time I may do it. But probably not in the next few weeks.

eku, 2018-01-31:
@tu500 it has been a long time since your last statement. Any chance to get a pull request from you?