ethersex/ethersex #413: make dependencies for meta_magic_scheduler

eku, 2015-06-12:
After ```make menuconfig``` the meta_magic_scheduler isn't rebuilt.

michaelb42, 2015-06-14:
This bug is not restricted to the new timer framework, it's a long unnoticed one also in the head branch.

meta.c and meta.h do not depend on meta.defines. Changing meta.defines via make menuconfig does not trigger a regeneration of these source files.

#414 fixes this one. Please test.

eku, 2015-06-15:
The timestamp ```built on <date><time>``` isn't updated on each new build.

eku, 2015-06-15:
Tested your fix, but doesn't work for me (gnu make 3.82).

michaelb42, 2015-06-15:
The timestamp mentioned above is defined in config.h and evaluated during compile time of ethersex.c - if this src is *not* rebuild on each and every change to config.h / autoconf.h something is horribly wrong with your setup.

Second - works for me :wink: Please provide a "make -d" log without / with the fix applied from a fresh clean build and a build where only a value_XXXX element is changed via menuconfig, e.g. the hostname.

eku, 2015-06-16:
If I change a source file (not ethesex.c), the timestamp is not updated. Since it says "built on" and not "configured on" the info is misleading.

closed 2016-02-07 by eku