ethersex/ethersex #421: ecmd_sender_state.h:32:5: error: expected specifier-qualifier-list before ‘client_return_text_callback_t’

macmattes, 2015-09-01:
In file included from ./meta.h:9:0,
                 from ./protocols/uip/uip.h:59,
                 from ./protocols/ecmd/sender/ecmd_sender_net.h:30,
                 from control6/control6.c:14:
./protocols/ecmd/sender/ecmd_sender_state.h:32:5: error: expected specifier-qualifier-list before ‘client_return_text_callback_t’

mit zbus enabled auf uart1 gibt es obigen fehler, sorry wenns zu knapp ist, ich kann es nicht besser beschreiben, so weit ich gesehen hab gab es im April ein update in der datei, seit dem oder danach ist der fehler reingekommen

eku, 2015-09-01:
git bisect ?

macmattes, 2015-09-01:
was? hab mir gerade mal ne frische git version gezogen und nur meine
configs zusammengestellt. kommt der gleiche fehler, gibt es da ne datei die
weiterhelfen könnte?
habe ethernet forwarding auf zbus und ecmd send tcp und udp, dann noch i2c,
onewire und dht
liegt anscheinend an control6, hab es mit einer leeren control6 versucht,
auch wieder der fehler
In file included from ./meta.h:9:0,
                 from ./protocols/uip/uip.h:59,
                 from ./protocols/ecmd/sender/ecmd_sender_net.h:30,
                 from control6/control6.c:14:
./protocols/ecmd/sender/ecmd_sender_state.h:32:5: error: expected
specifier-qualifier-list before ‘client_return_text_callback_t’
In file included from control6/control6.c:52:0:
./hardware/lcd/hr20.h:27:1: error: expected expression before ‘void’
und ein fehler mehr, ohne control6 rennt er anstandslos durch

eku, 2015-09-02:
Wenn Du schon nicht gewillt bist, mittels 'git bisect' den verursachenden Commit zu bestimmen, dann häng wenigstens Deine aufoconf.h an.

macmattes, 2015-09-02:
ah so, wusste nicht was gewiollt war, bin nicht so fit sorry.
gibt alles heute abend wenn ich zuhause bin

macmattes, 2015-09-02:
hallo, hab nun null ahnung wie man bisect bedient start/good/bad/run? Hab
mal was probiert
start:
Already on 'master'
git bisect good
git bisect bad
1f115abc3c215ec872de3a50261a6c1ad1076c00 was both good and bad

wie gesagt es rennt durch wenn control6 ausgeschaltet ist und nicht wenn
control an ist

eku, 2015-09-02:
Was steht in deinen C6 script?

macmattes, 2015-09-02:
nur noch 
CONTROL_START

CONTROL_END

hab es sogar mal mit einer ganz leeren datei versucht, die geht auch nicht

eku, 2015-09-02:
Ich habe auch die HR20 Config ausgewählt unde Send ECMD hinzugenommen und es übersetzt. Bitte Dein .config anhängen.

macmattes, 2015-09-02:
die config

macmattes, 2015-09-02:
komisch ich hab sie in der mail angehangen

eku, 2015-09-08:
Welche Email? Du kannst Dateien hier ans Issue anhängen ("select item") oder nutze gist.gitgub.com.

macmattes, 2015-09-08:
die config lässt sich hier nicht hochladen, es wird ein bild erwartet, hab mal gist versucht, hoffe das war so richtig. ich hatte auf ein mail dass ich von Github bekommen hatte mit der config geantwortet, text steht oben aber der anhang kam wohl nicht durch

https://gist.github.com/macmattes/6572cd663b157dba0081#file-config

eku, 2015-09-24:
@macmattes: bitte um Rückmeldnung, ob der Patch das Problem bei Dir behebt.

macmattes, 2015-09-24:
Hallo
ja , läuft durch, danke

closed 2015-09-24 by eku