ethersex/ethersex #433: YPort drops packets when simultaneously receiving and sending

kilroy42, 2015-10-24:
I wrote a simple test which justs sends 4 bytes and then receives 4 bytes. It will do this twice, after this, the last sent packet is dismissed by EtherSex (but acked according to Wireshark!). You need to wait around 40msec to send another packet - if you do not, all packets are dismissed.

This is due to a simple bug in yport_net.c:

```diff
diff --git a/protocols/yport/yport_net.c b/protocols/yport/yport_net.c
index 6eb27a8..a1a10cf 100755
--- a/protocols/yport/yport_net.c
+++ b/protocols/yport/yport_net.c
@@ -85,8 +85,10 @@ yport_net_main(void)
     if (yport_conn == uip_conn)
       yport_conn = NULL;
   }
-  else if (uip_newdata())
+  
+  if (uip_newdata())
   {
+
     if (uip_len <= YPORT_BUFFER_LEN &&
         yport_rxstart(uip_appdata, uip_len) != 0)
     {
```
Explanation: When you just sent a packet and get the ACK and newdata, you will only run into the ACK path, and newdata is dropped. Removing the else takes care of that.

With a short grep I found more (possibly) errors:

```grep
protocols/ems/ems_net.c:  } else if (uip_newdata()) {
protocols/modbus/modbus_net.c:  } else if (uip_newdata()) {
protocols/mqtt/mqtt.c:  else if (uip_newdata() && uip_len) {
protocols/irc/irc.c:    else if (uip_newdata() && uip_len) {
protocols/bsbport/bsbport_net.c:  else if (uip_newdata())
```

eku, 2015-10-27:
The issue should stay open as long as all the mentioned occurences are not valided.

eku, 2026-08-20:
## The Bug
When a TCP connection receives an ACK and new data in the same uIP poll cycle, the else if (uip_newdata()) structure causes the new data handler to be skipped if the ACK handler runs first.
Root cause: Chained else if statements for uip_acked() followed by uip_newdata() prevent both from executing when both flags are set.

##  Can It Still Happen?
NO - The fix has already been applied to yport_net.c. The else if (uip_newdata()) was changed to if (uip_newdata()), allowing both ACK and new data to be processed independently when they occur on the same connection.

Scenario that previously failed but now works:
 1. YPort sends a packet
 2. Server sends ACK + new data in same TCP segment
 3. uIP calls yport_net_main() with both uip_acked() and uip_newdata() true
 4. Before fix: Only ACK processed, new data ignored
 5. After fix: Both ACK and new data processed correctly

Status: ✅ FIXED - YPort no longer drops packets in this scenario.

closed 2026-08-20 by eku