ethersex/ethersex #472: Very strange DHT22 bug in menuconfig

tobox, 2016-10-13:
I tried to connect a DHT22 to my AVR NetIO, without success. It compiled fine, but reading values from the sensor only resulted in temp 0.0 and humidity 0.0 or 2.0.

After many hours of debugging, I found (part of) the problem. When I run "make menuconfig" and go to "I/O", "Humidity & temperature sensors", "DHT 11/22", Sensor type was set to DHT11. When I hit enter to change the sensor type, it opened the pin configuration in emacs (that should only happen if I hit enter on "edit pin configuration"). So I thought that there was only one sensor type available, and DHT11 whould be compatible with DHT22.

After changing DHT11 to DHT22 manually in the .config and running "make menuconfig" again, the sensor type is properly displayed as "DHT22/AM2302". If I try to change the sensor type, the correct menu opens and offers "DHT11" and "DHT22/AM2302". If I select DHT11, the erraneous behaviour described above occurs and I cannot change the value back to DHT22. 

I have absolutely no clue if this is a ethersex problem or some kind of toolchain problem. I am running debian testing on amd64.

With a fresh checkout of the ethersex source, I was unable to reproduce the problem. But I am not aware of any changes that I made that could have resulted in the observed behaviour.

michaelb42, 2016-10-16:
I can confirm this bug. The behaviour may be reproduced - including, but not limited to -  by

1. do a fresh clone
2. choose Net-IO w/ add-on  (in my test case)
3. enable DHT 11/22 and enter the menu
4. select DHT22/AM2302 from the choice end leave the coice with enter
5. now select DHT11 again from the choice and enter

... subsequent hits on enter will not present the sensor type choice but the pin configuration editor.

michaelb42, 2016-10-23:
Merged #473

closed 2016-10-23 by michaelb42

florian-asche, 2016-10-24:
I whould also add the idea, if we can add the setting for each sensor to the configuration file. If you want to use a mixed setup, DHT11 and DHT22, thats with the configuration at this moment not possible.

eku, 2016-10-25:
@florian-asche the code does not support mixed type sensors.