ethersex/ethersex #475: DHCP renew implemented?

tobox, 2016-11-15:
My Net-IOs always disappeared from the LAN after approximately 10 days, and I think I found the root cause of the problem. I use DHCP, and my router (Fritz!Box 7490) discards the DNS-Mappings when the DHCP lease time expires. The Net-IOs can still be reached by the IP address, but not by the DNS name. 

I think the behaviour of the router is correct, so I looked into the DHCP code of ethersex and could not find any DHCP renew code. Is it correct, that it is just not implemented? Any specific reason why it is not implemented?

Regards

eku, 2016-11-15:
Yes, renewing is not implemented yet. PR are very welcome.

eku, 2019-06-23:
The Leasetime is read from the response of the DHCP server and even stored to be ignored.

PR very welcome!

eku, 2019-06-23:
DHCP implementation on Contiki does renewing:
https://github.com/contiki-os/contiki/blob/master/core/net/ip/dhcpc.c#L352

closed 2026-08-20 by eku