ethersex/ethersex #487: memory leak / race condition in httplog

ferbar, 2017-03-26:
httplog stops working if requests sent too fast

fixed in #484

eku, 2017-03-27:
Since you tried to fix the issue could you please explain what makes you thinking that Ethersex has a memory leak? Where ist the leak (file/line number)?

ferbar, 2017-04-12:
This is the race condition if 2 subsequent calls to httplog() are made

1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 
-- | - | - | - | - | - | -  | - 
first call to httplog("some text") | httplog_buffer_empty()=> true + DNS query + connect | httplog_net_main() STATE=connected, sending http request | httplog_net_main() STATE=ACK, uip_close() free(httplog_tmp_buf) |  httplog_net_main() STATE=CLOSE, uip_close() free(httplog_tmp_buf) (again) | httplog_net_main() STATE=ABORT, uip_close() free(httplog_tmp_buf) (again) | 
2nd call to httplog() | | | httplog_buffer_empty()=> true + DNS query + connect | httplog_net_main() STATE=connected, sending http request | httplog_net_main() STATE=ACK, uip_close() free(httplog_tmp_buf) |  httplog_net_main() STATE=CLOSE, uip_close() free(httplog_tmp_buf) (again) | httplog_net_main() STATE=ABORT, uip_close() free(httplog_tmp_buf) (again) | 

In Step 5 of the 2nd request no data is sent because the first request does a second free(httplog_tmp_buf) and frees the 2nd requests data.

The ABORT - step is a bug in uip.c, see #486 

This is the memory leak:
Sometimes httplog_tmp_buf is never freed, so httplog stops working after a random time period. I think that this can be caused by a bad internet connction, the remote server is down, the remote server is sending a reply rather than an empty page or the stack is overwriting the heap area.

I'm still investigating this.

commit 5504f068 (2019-11-07): Merge pull request #547 from eku/fix_487_httplog

Fix #487
commit 8c590001 (2019-11-07): Fix #487
Store requests in queue.
closed 2019-11-07 by eku