ethersex/ethersex #502: ethersex/watchasync counter overflow?

ulrich-knur, 2017-10-26:
I'm using AVR NET-IO with ethersex and watchasync to feed the pulses of an electricity meter to volkszaehler.  Electricity meter gives 10000 pulses per kwh. watchasync is configured to include unix timestamp, resolution 32 seconds,  summarize events,  16 bit counterrange, buffersize 64, no polling.
Works fine as long as electric power is lower than appr. 2,8kw. Above this value, volkszaehler frontend shows  kw values starting up from zero again. Seems to me, that watchasync uses a 8 bit integer (maximum 255) to sum up the pulses within the time period of 32 seconds: (255*3600)/(32*10000)=2,86875kw is the maximum kw value that can fit in 8 bits within 32 seconds.
Regards Uli

eku, 2017-10-31:
Use config option ```Counterrange``` to increase countable range. Default value is 16 bit.

closed 2017-10-31 by eku

ulrich-knur, 2017-10-31:
Thanks for your response. Counterrange has already been set to 16 (bits). Do  i need more (32 bits) for the counter to exceed 255?