ethersex/ethersex #503: Changeset b827d22 breaks menuconfig menus for usart

stefan-muc, 2017-10-31:
git checkout ca6f98d
make menuconfig
Go to Protocols ---> ECMD support ---> 
Enable USART (if not enabled)
Press [Enter] on "Ecmd usart select"
New dialogue for selecting usart appears

git checkout b827d22
do the same as above
No dialogue appears.

`./MCradiolists: line 12: syntax error near unexpected token `('
./MCradiolists: line 12: `      function  ()            { l_choice '  Ecmd usart select' "  ECMD_SERIAL_USART_USART_" "" "" ;}'
`

michaelb42, 2017-10-31:
@eku It seems that the promicro PR #500 breaks the USART handling

eku, 2017-11-01:
@michaelb42 works perfectly on Fedora 26. Could you please provide the temporary generated files. thx

michaelb42, 2017-11-01:
@stefan-muc could you please provide the files in question to @eku?

stefan-muc, 2017-11-01:
Asked @michaelb42 on details what "the temporary generated files" are.

Here there are:
[MCmenu.txt](https://github.com/ethersex/ethersex/files/1435623/MCmenu.txt)
[MCradiolists.txt](https://github.com/ethersex/ethersex/files/1435624/MCradiolists.txt)

PR #507 (2017-11-06): Fix for issue #503
avrlibc defines UDR without number suffix for some MCUs which breaks USART detection.
fixes #503
closed 2017-11-06 by michaelb42