ethersex/ethersex #504: 1wire naming support breaks board

stefan-muc, 2017-10-31:
Using 0c2e083

Use this config and compile for Pollin netio:
[config_2017-10-31_1wire.txt](https://github.com/ethersex/ethersex/files/1431678/config_2017-10-31_1wire.txt)

flash it
Ping board by IP 192.168.0.100 - success

Now enable I/O ---> Onewire support ---> Onewire naming support
make and flash
Ping board -> timeout. Board doesn't work anymore.

eku, 2017-11-07:
@stefan-muc does it work if you turn of ```EEPROM memory is preserved throug the Chip Erase``` in the AVR fuses, i.e. EEPROM is always erased when you flash a new ethersex?

stefan-muc, 2017-11-15:
@eku We already tried that before reporting the bug.
I tried it again just now:
* Updated to most recent commit 7ed1212
* make with my current config
* unset EESAVE fuse, program and verify; fuse is not set
* program compiled ethersex.hex: It works as expected (board is pingable)

* enable onwire naming support, compile and program => board is not pingable
* disable onewire naming support, compile and program => board is pingable again

michaelb42, 2017-11-15:
Tested with current head https://github.com/ethersex/ethersex/commit/7ed12121640bf354142f2a6454fb502426e58845, config file from initial report and pinning file from reporter,
changed:
- MCU to 1284p since I had no 32 available
- Freq to 20 MHz (Crystal on board)
- IP-Address.

1st run, without 1w naming: ping works as expected
2nd run, activated 1w naming only: pink still works as expected

not reproducible here.

stefan-muc, 2017-11-15:
Tried another ATmega32, same results as reported :(

michaelb42, 2017-11-16:
Extract from test results by djmaster (http://wiki.senseye.org/adir/stef.txt, via irc #ethersex):
--
```
http://192.168.123.18/ecmd?1w name set 0 10cf2e88020800e6 test

10cf2e88020800e6	test	21.9	0
109f548802080018		22.6	0
OK

- Version 0c2e083
- statt PD6 nun PD4
- etherrape config, nicht netio
- mega644p
- 20MHz
- VFS abgeschaltet

root@avr /home/djmaster/avr/ethersex_beta/ethersex_StefanMUC/0c2e083 # ping 192.168.123.18

PING 192.168.123.18 (192.168.123.18) 56(84) bytes of data.
64 bytes from 192.168.123.18: icmp_seq=1 ttl=64 time=16.0 ms
64 bytes from 192.168.123.18: icmp_seq=2 ttl=64 time=8.16 ms
64 bytes from 192.168.123.18: icmp_seq=3 ttl=64 time=14.0 ms
```
--
- GLEICHE Firmware wie oben mit DEBUG ON

```
D: ethersex  built on Nov 16 2017 16:09:07 (Debug mode)
D: disabling watchdog
D: enc28j60 revision 0x6
D: mac: 02:ca:fe:64:2b:1e
D: net: got link!
D: OW-POLL: starting discovery
D: OW-POLL: discovered device 10cf2e88020800e6
D: OW-POLL: stored new sensor in pos 0
D: OW-POLL: discovered device 109f548802080018
D: OW-POLL: stored new sensor in pos 1
D: OW-POLL: start conversion on all sensors
D: OW-POLL: temperature: 22.1Â°C on device 10cf2e88020800e6 0
D: OW-POLL: temperature: 22.8Â°C on device 109f548802080018 0
D: OW-POLL: start conversion on all sensors
D: OW-POLL: temperature: 22.1Â°C on device 10cf2e88020800e6 0
D: OW-POLL: temperature: 22.8Â°C on device 109f548802080018 0
```
--

michaelb42, 2017-11-16:
@stefan-muc not able to reproduce this bug. Closing this one.
Feel free to re-open this bug, if you can provide more info and a reproducible case.

closed 2017-11-16 by michaelb42

stefan-muc, 2017-11-17:
Compiled with same config as @michaelb42 used, got different binary size (because of different toolchain) but it worked at his board.

Tried to disable VFS support, same symptoms.

Enabled Heartbeat LED. It blinks without 1wire naming support, but doesn't work with 1wire naming support.

Can someone with ATMega32 try to reproduce this bug? From what I tested I expect this to be a model specific bug as it works with other ones.