ethersex/ethersex #509: Can't make menuconfig - .config file missing / ncurses

TheSmartGerman, 2017-11-10:
I did a new Ubuntu/ethersex installation according to the tutorial http://www.ethersex.de/index.php/Quick_Start_Guide.

When I try to make menuconifg, I'll get a lot of errors .config file is missing. At the end, I'll also tells me unable to find Ncurses libraries. 

I checked the Ncurses libraries i tried this: https://stackoverflow.com/questions/23050188/cant-make-menuconfig

but no success...

TheSmartGerman, 2017-11-10:
Ok, got it!
Missing: Native GCC-Compiler (for the lxdialog utility required by make menuconfig) 

The guide should be adjusted to this: `apt-get install **gcc** gcc-avr avr-libc binutils-avr m4 gawk libncurses5-dev make dialog git-core avrdude`

closed 2017-11-10 by TheSmartGerman

eku, 2017-11-16:
@TheSmartGerman why don't you just add the missing information to the wiki?

TheSmartGerman, 2017-11-24:
I don't have a Login. The Login page don't give you the Option to create a
new user. Only as a user I'm able to edit the wiki

TheSmartGerman, 2017-12-08:
Wer verwaltet das Wiki? Ich würde mich da gerne ein wenig einbringen. Aber
es bit keine Seite zum einen neuen Account zu erstellen...

2017-11-24 11:59 GMT+01:00 Benedikt Geiger <benegei@gmail.com>:

eku, 2017-12-14:
@TheSmartGerman wiki account created. You should have received an email with login credentials.

TheSmartGerman, 2017-12-14:
Ok thx