ethersex/ethersex #516: menuconfig & make & i2c on AVR-NET-IO w/addon and maybe others

rd65, 2018-01-26:
by choosing i2c slave code, the config System try to include i2c master files... whitch went wrong...

avr-gcc -Wall -W -Wno-unused-parameter -Wno-sign-compare -Wno-char-subscripts -g -Os -std=gnu99 -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -mcall-prologues -fshort-enums -fno-strict-aliasing -Iprotocols/usb/usbdrv -Iprotocols/usb -DAVR_BUILD -DF_CPU=16000000UL -mmcu=atmega1284p -I.  -c -o hardware/i2c/master/i2c_master.o hardware/i2c/master/i2c_master.c
hardware/i2c/master/i2c_master.c: In function ‘i2c_master_init’:
hardware/i2c/master/i2c_master.c:31:30: error: ‘CONF_I2C_BAUD’ undeclared (first use in this function)
     TWBR = ((F_CPU / 1000) / CONF_I2C_BAUD - 16) / 2;  // Frequenz setzen
                              ^
hardware/i2c/master/i2c_master.c:31:30: note: each undeclared identifier is reported only once for each function it appears in
**<eingebaut>: die Regel für Ziel „hardware/i2c/master/i2c_master.o“ scheiterte**
make: *** [hardware/i2c/master/i2c_master.o] Fehler 1

-----------------------------------

cat .config|grep I2C
# I2C_INLINE_SUPPORT is not set
CONF_HD44780_CONNECTION=HD44780_I2CSUPPORT
HD44780_I2CSUPPORT=y
CONF_HD44780_I2C_PORTEXP=HD44780_I2C_PCF8574
HD44780_I2C_PCF8574=y
# HD44780_I2C_MCP23017 is not set
**I2C_MASTER_SUPPORT=y**
I2C_PCF8574X_SUPPORT=y
# DEBUG_HD44780_I2C is not set
# GLCD_I2C_SUPPORT is not set
# GLCD_MONO_LD7032_60x32_I2C is not set
# GLCD_MONO_SH1106_128x64_I2C is not set
# GLCD_MONO_SSD1305_128x32_I2C is not set
# GLCD_MONO_SSD1306_64x32_I2C is not set
# GLCD_MONO_SSD1306_64x48_I2C is not set
# GLCD_MONO_SSD1306_96x16_I2C is not set
# GLCD_MONO_SSD1306_128x64_I2C is not set
# GLCD_MONO_SSD1309_128x64_I2C is not set
# GLCD_MONO_SSD1325_128x64_I2C is not set
# GLCD_MONO_SSD1326_256x32_ER_I2C is not set
# GLCD_MONO_SSD1327_96x96_I2C is not set
# GLCD_MONO_ST75256_JLX172104_I2C is not set
# GLCD_MONO_ST75256_JLX256128_I2C is not set
# GLCD_MONO_ST75256_JLX25664_I2C is not set
# GLCD_MONO_ST7588_JLX12864_I2C is not set
# GLCD_MONO_UC1601_128x32_I2C is not set
# GLCD_MONO_UC1604_JLX19264_I2C is not set
# GLCD_MONO_UC1608_240x128_I2C is not set
# GLCD_MONO_UC1608_ERC240120_I2C is not set
# GLCD_MONO_UC1608_ERC24064_I2C is not set
# GLCD_MONO_UC1610_EA_DOGXL160_I2C is not set
# GLCD_MONO_UC1611_EA_DOGM240_I2C is not set
# GLCD_MONO_UC1611_EA_DOGXL240_I2C is not set
# GLCD_MONO_UC1611_EW50850_I2C is not set
# I2C
# I2C_MASTER_SUPPORT is not set
# I2C_DETECT_SUPPORT is not set
# I2C_GENERIC_SUPPORT is not set
# I2C_24CXX_SUPPORT is not set
# I2C_LM75_SUPPORT is not set
# I2C_TMP175_SUPPORT is not set
# I2C_DS1631_SUPPORT is not set
# I2C_DS13X7_SUPPORT is not set
# I2C_PCF8583_SUPPORT is not set
# I2C_TSL2550_SUPPORT is not set
# I2C_TSL2561_SUPPORT is not set
# I2C_BMP085_SUPPORT is not set
# I2C_PCA9531_SUPPORT is not set
# I2C_PCA9685_SUPPORT is not set
# I2C_PCF8574X_SUPPORT is not set
# I2C_PCA9555_SUPPORT is not set
# I2C_MAX7311_SUPPORT is not set
# I2C_MCP23017_SUPPORT is not set
# I2C_UDP_SUPPORT is not set
# DEBUG_I2C is not set
**I2C_SLAVE_SUPPORT=y**
# DEBUG_I2C_SLAVE is not set
# ECMD_SERIAL_I2C_SUPPORT is not set

---------------------------

The Problem....
On AVR-NET-IO w/addon and maybe others you cant use I2C as slave without disabling the Addon I2C Display but its selectable in Menuconfig... 
Solution:
1. predefine all i2C Master activity as I2C Master in Menuconfig and give a hint in help files...
2. invent a I2C Lib witch is Master AND Slave capable and switches on request or always Slave, Master on request

michaelb42, 2018-02-02:
ethersex i2c slave code is a mess, maybe we should disable it completely?

rd65, 2018-02-02:
do you mean the i2c slave c code or the m4 macros as messy code?
the bug described on top depends on m4 macros/menuconfig... and a simple i2c slave (like arduino) is a tiny irq service in c with registers... i don't complain about the c i2c slave...

michaelb42, 2018-02-02:
The i2c slave code. Do you actually have a working real world setup using the existing i2c slave stuff?

rd65, 2018-02-03:
Today no, but maybe tomorrow. The Point is... most People use i2c/twi as Master to poll simple Sensors. (cpu)#(sensors) The  Slave Code is only necessary if you use a CPU Board as a Sub Bord aka Sensor Multiplexer in a more complex Board Architecture. (cpu)->(cpu)#(sensors). But there is no way to do interrupts requested by Slaves or Sub Boards. But CPU Boards can use each other as slaves if they are always in Slave/Listen Mode and switch only to Master/Sender on request. Its similar to a local Ethernet in halfduplex mode without the need of complex Network Hardware. This enables IRQ in both directions too. (sensors)#(cpu)<->(cpu)#(sensors)
The conclusion:
1. Slave code is not necessary on simple i2c busses, but use only 10% of the capabilities of i2c
2. a better c Slave/Master code is necessary on complex i2c busses to use the hole i2c thing.
In my opinion, its not a good idea to reduce software capabilities without a big reason. And missing Knowledge is never a reason. The question is... do the actual slave code the job as it is designed? If yes.. no reason to remove... if no.. a reason to do rework on it...
For real Master/Slave Multiplex mode we need some more Brainwork! But there are running Examples in the Arduino World.

michaelb42, 2018-02-03:
1. There's plenty of brain around here but in general it's used for more meaningful things.
2. If you know it all (TM) then do us the favour and enlighten us with a PR instead of spamming a github issue with a discussion. There's a mailing list.
3. And please don't forget - we don't do Arduino here.

rd65, 2018-02-03:
1. 2. and 3. ... I ask for fixing the M4 Code! Nothing else! You discuss about Slavecode!

michaelb42, 2018-02-04:
e6 supports either master or slave but not both on one device. It's a reasonable design decision.
slave support already depends on not master, you cannot select both. A proper configuration for the i2c slave (for the time being IMO only example) code compiles fine here.

No bug.

closed 2018-02-04 by michaelb42