ethersex/ethersex #518: menuconfig & make & IRMP on AVR-NET-IO w/addon

rd65, 2018-02-01:
The IRMP config for AVR-NET-IO w/addon seem to be very broken.

```
ifdef(`conf_IRMP', `dnl
dnl pin(IRMP_RX, PD3) the 
#undef IRMP_USE_TIMER2
#define IRMP_RX_LOW_ACTIVE
#undef IRMP_RX_LED_LOW_ACTIVE
dnl pin(STATUSLED_IRMP_RX, PC3, OUTPUT)
pin(IRMP_TX, PD7) dnl OC2/OC2A
#undef IRMP_TX_LED_LOW_ACTIVE
dnl pin(STATUSLED_IRMP_TX, PC2, OUTPUT)
')
```
PD3(INT1) is the NIRQ port on RFM12 !
PD7 is the CARDIN on SD Slot...
PC2/3 as status LEDs are routed to the 25 pin Bus ...

But the TSOP Out is PB0 ... and not named...

rd65, 2018-02-02:
i found a correct pinning in arduino format for AVR-NET-IO w/addon on:
[https://github.com/cniweb/avr-netino]
i hope, this will help fixing pin issues. i' not familiar with m4.

```
#ifndef AVR_NET_IO 
#define AVR_NET_IO	0x20150214	/* Date YYYMMDD */
#endif

#ifdef pinDef
/***********************************************************
 * mapping from pin number to :
 * Port,Bit,Timer(PWM),spec.function,usage,pcint,...
 ***********************************************************/
/* ATmega with 4 Ports (m32/m644) = AVR-Net-IO */
/* 0 */
pinDef( C, 0 , NOT_ON_TIMER,SCL,J3_2,16)
pinDef( C, 1 , NOT_ON_TIMER,SDA,J3_3,17)
pinDef( C, 2 , NOT_ON_TIMER,TCK,J3_4,18)
pinDef( C, 3 , NOT_ON_TIMER,TMS,J3_5,19)
pinDef( C, 4 , NOT_ON_TIMER,TDO,J3_6,20)
pinDef( C, 5 , NOT_ON_TIMER,TDI,J3_7,21)
pinDef( C, 6 , NOT_ON_TIMER,TOSC1,J3_8,22)
pinDef( C, 7 , NOT_ON_TIMER,TOSC2,J3_9,23)
/* 8 */
pinDef( A, 0 , NOT_ON_TIMER,ADC0,A0,0)
pinDef( A, 1 , NOT_ON_TIMER,ADC1,A1,1)
pinDef( A, 2 , NOT_ON_TIMER,ADC2,A2,2)
pinDef( A, 3 , NOT_ON_TIMER,ADC3,A3,3)
pinDef( A, 4 , NOT_ON_TIMER,ADC4,ADC_1,4)
pinDef( A, 5 , NOT_ON_TIMER,ADC5,ADC_2,5)
pinDef( A, 6 , NOT_ON_TIMER,ADC6,ADC_3,6)
pinDef( A, 7 , NOT_ON_TIMER,ADC7,ADC_4,7)
/* 16 */
pinDef( D, 0 , NOT_ON_TIMER,RXD,RS232_RxD,24)
pinDef( D, 1 , NOT_ON_TIMER,TXD,RS232_TxD,25)
pinDef( D, 2 , NOT_ON_TIMER,INT0,LED_1,26)
pinDef( D, 3 , NOT_ON_TIMER,INT1,RFM12_IRQ,27)
pinDef( D, 4 , TIMER1B,OC1B,LED_2,28)
pinDef( D, 5 , TIMER1A,OC1A,RFM12_CS,29)
#ifdef COM21
pinDef( D, 6 , NOT_ON_TIMER,ICP1,LED_3,30)
pinDef( D, 7 , TIMER2,OC2,SDcard_INS,31)
#else
pinDef( D, 6 , TIMER2B,OC2B,LED_3,30)
pinDef( D, 7 , TIMER2A,OC2A,SDcard_INS,31)
#endif
/* 24 */
pinDef( B, 0 , NOT_ON_TIMER,T0,IR_Rx,8)
pinDef( B, 1 , NOT_ON_TIMER,T1,JUMP_PROG,9)
pinDef( B, 2 , NOT_ON_TIMER,INT2,ENC28J60_IRQ,10)
pinDef( B, 3 , TIMER0A,OC0,SDcard_CS,11)
#ifdef COM0B1
pinDef( B, 4 , TIMER0B,SS,ENC28J60_CS,12)
#else
pinDef( B, 4 , NOT_ON_TIMER,SS,ENC28J60_CS,12)
#endif
pinDef( B, 5 , NOT_ON_TIMER,MOSI,SPI_MOSI,13)
pinDef( B, 6 , NOT_ON_TIMER,MISO,SPI_MISO,14)
pinDef( B, 7 , NOT_ON_TIMER,SCK,SPI_SCK,15)
#endif /* pinDef */
```

eku, 2018-02-03:
@rd65  A PR is very welcome.

rd65, 2018-02-03:
i'will try it but i'm not familiar with m4 Macros. Need some Time for testing.

closed 2018-02-04 by rd65