ethersex/ethersex #526: ATmega32: Network Buffer Size limited to 500 bytes

ferbar, 2018-03-27:
### Expected behavior
make menuconfig -> Network -> Network Buffer Size
enter 600
Network Buffer Size stays 600

A FRITZ!Box produces DHCP packages with a size of 590 bytes, so this value has to be increased to get DHCP working on an ATmega32 with a fritz box.

### Actual behavior
Value is reduced to 500 in https://github.com/ethersex/ethersex/blob/master/config.in#L198

I don't see a technical reason not to allow for exampe 600 bytes, what is the limit for the ATmega16.

### Ethersex version
d883c87

### Hardware
Target MCU ATmega32

eku, 2018-03-28:
See also https://github.com/sde1000/NanodeUIP/issues/4

closed 2018-04-19 by eku

ferbar, 2019-02-03:
@eku: The original FRITZ!Box firmware (4.83) does not anwer to BOOTP requests, so sadly BOOTP is not an option. The FRITZ!Box adds 240 bytes \0 padding to responses. (....)

You're right, i would not change the default network buffer size. I would only change the maximum allowed network size (NET_MAX_FRAME_LENGTH) to easily allow a bigger buffer size via make menuconfig. 

At least in my case increasing the buffersize was a simple fix because I don't need much ram.
DHCP + 600 bytes network buffer:
Program (.text + .data) : 30504 bytes
Data (.data + .bss)     : 1626 bytes