ethersex/ethersex #527: Introduction of -flto option

LineF, 2018-04-14:
With the introduction of the -flto option to gcc and ld my code size rises from about 38KB to 52KB!
Additionally I get link time warnings:

avr-gcc -Wall -W -Wno-unused-parameter -Wno-sign-compare -Wno-char-subscripts -g -Os -std=gnu99 -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -flto -mcall-prologues -fshort-enums -fno-strict-aliasing -Iprotocols/usb/usbdrv -Iprotocols/usb -DAVR_BUILD -DF_CPU=20000000UL -mmcu=atmega1284p -I.  -c -o meta.o meta.c
avr-gcc -Wall -W -Wno-unused-parameter -Wno-sign-compare -Wno-char-subscripts -g -Os -std=gnu99 -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -flto -mcall-prologues -fshort-enums -fno-strict-aliasing -Iprotocols/usb/usbdrv -Iprotocols/usb -DAVR_BUILD -DF_CPU=20000000UL -mmcu=atmega1284p -I.  -c -o core/periodic_milliticks.o core/periodic_milliticks.c
avr-gcc -Wall -W -Wno-unused-parameter -Wno-sign-compare -Wno-char-subscripts -g -Os -std=gnu99 -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -flto -mcall-prologues -fshort-enums -fno-strict-aliasing -Iprotocols/usb/usbdrv -Iprotocols/usb -DAVR_BUILD -DF_CPU=20000000UL -mmcu=atmega1284p -I.  -c version.c
avr-gcc -mmcu=atmega1284p -Wl,--gc-sections,--relax -flto -o ethersex ethersex.o core/eeprom.o core/util/fixedpoint.o core/util/string_parsing.o core/util/byte2bin.o core/util/byte2hex.o core/util/app_crc.o core/queue/queue.o network.o core/periodic.o core/spi.o core/heartbeat.o hardware/adc/temp2text.o hardware/avr/dump_ecmd.o hardware/avr/fuse_ecmd.o hardware/avr/hostname_ecmd.o hardware/avr/reset_ecmd.o hardware/avr/bootloader_ecmd.o hardware/ethernet/ethernet_config.o hardware/ethernet/enc28j60.o hardware/ethernet/enc28j60_process.o hardware/ethernet/enc28j60_transmit.o hardware/dht/dht.o hardware/i2c/master/i2c_master.o hardware/i2c/master/i2c_bmp085.o hardware/lcd/hd44780_core.o hardware/lcd/hd44780.o hardware/onewire/onewire.o protocols/bootp/bootp_net.o protocols/bootp/bootp.o protocols/ems/ems.o protocols/ems/ems_net.o protocols/ems/ems_uart.o protocols/ems/ems_proto.o protocols/syslog/syslog_net.o protocols/syslog/syslog.o protocols/syslog/syslog_debug.o protocols/uip/uip.o protocols/uip/uip_multi.o protocols/uip/uip_router.o protocols/uip/parse.o protocols/uip/uip_arp.o protocols/uip/check_cache.o protocols/ecmd/parser.o protocols/ecmd/via_tcp/ecmd_net.o protocols/ecmd/via_usart/ecmd_usart.o services/alarm/alarm.o services/clock/clock.o services/clock/clock_lib.o services/ntp/ntp.o core/periodic_ecmd.o core/portio/ecmd_simple.o hardware/ethernet/ethernet_ecmd.o hardware/dht/dht_ecmd.o hardware/i2c/master/i2c_master_ecmd.o hardware/i2c/master/i2c_bmp085_ecmd.o hardware/lcd/lcd_ecmd.o hardware/onewire/onewire_ecmd.o hardware/radio/bluetooth/bt_ecmd.o protocols/ems/ems_ecmd.o protocols/uip/ipstats.o protocols/uip/ecmd.o services/alarm/ecmd.o services/clock/clock_ecmd_date.o services/clock/clock_ecmd_time.o services/clock/uptime_ecmd.o services/ntp/ntp_ecmd.o meta.o core/periodic_milliticks.o  version.o -lm -lc # Pixie Dust!!! (Bug in avr-binutils)
In function ‘__vector_15’:
core/periodic_milliticks.c:32:1: warning: ‘_vector_15’ appears to be a misspelled signal handler [enabled by default]
 ISR(PERIODIC_VECTOR_OVERFLOW)
 ^
In function ‘__vector_20’:
protocols/ecmd/via_usart/ecmd_usart.c:102:1: warning: ‘_vector_20’ appears to be a misspelled signal handler [enabled by default]
 ISR(usart(USART,_RX_vect))
 ^
In function ‘__vector_22’:
protocols/ecmd/via_usart/ecmd_usart.c:133:1: warning: ‘_vector_22’ appears to be a misspelled signal handler [enabled by default]
 ISR(usart(USART,_TX_vect))
 ^
In function ‘__vector_9’:
protocols/ems/ems_uart.c:197:1: warning: ‘_vector_9’ appears to be a misspelled signal handler [enabled by default]
 ISR(TC2_VECTOR_COMPARE)
 ^
In function ‘__vector_30’:
protocols/ems/ems_uart.c:208:1: warning: ‘_vector_30’ appears to be a misspelled signal handler [enabled by default]
 ISR(usart(USART, _TX_vect))
 ^
In function ‘__vector_29’:
protocols/ems/ems_uart.c:214:1: warning: ‘_vector_29’ appears to be a misspelled signal handler [enabled by default]
 ISR(usart(USART,_UDRE_vect))
 ^
In function ‘__vector_28’:
protocols/ems/ems_uart.c:260:1: warning: ‘_vector_28’ appears to be a misspelled signal handler [enabled by default]
 ISR(usart(USART,_RX_vect))
 ^
avr-objcopy -O ihex -R .eeprom ethersex ethersex.hex
avr-objcopy -O binary -R .eeprom ethersex ethersex.bin
avr-objdump -h -S ethersex > ethersex.lss
=======The ethersex project========
Compiled for: atmega1284p at 20000000Hz
Imagesize: 53978/131072 bytes (41.18%)
  [============------------------]

Program (.text + .data) : 53978 bytes
Data (.data + .bss)     : 3367 bytes

===================================

eku, 2018-04-18:
Please supply GCC and Binutils version you are using.

LineF, 2018-04-18:
"apt list" gives me following information:

avr-libc/oldstable,now 1:1.8.0+Atmel3.4.4-1 all [installed]
avrdude/oldstable,now 6.1-2 armhf [installed]
binutils-avr/oldstable,now 2.24+Atmel3.4.4-1 armhf [installed]
gcc-avr/oldstable,now 1:4.8.1+Atmel3.4.4-2 armhf [installed]

"avr-gcc -v" gives me version 4.8.1
Regarding "apt list" I think binutils is version 2.24

All of that is running on a cubietruck with armbian.
"apt-get install" tells me that these versions are up to date...

eku, 2018-04-19:
Works perfectly with GCC 5.4.0/Binutils 2.28 and GCC  7.3.0/Binutils 2.30.
If you can't update you toolchain, please remove option ```-flto``` in ``` scripts/defaults.mk``` of your local branch.

closed 2018-04-19 by eku

LineF, 2018-04-26:
I have built the avr toolchain, gcc and libc with the current version from scratch and compiled ethersex again with the -flto option and it worked now.
Thanks for the tip!

Martin