ethersex/ethersex #536: MQTT goes wild after server restart

stefan-muc, 2018-10-03:
### Expected behavior
After a MQTT server restart, ethersex shall simply reconnect and resume operation

### Actual behavior
After a MQTT server restart and a little wait time (< 60 seconds - supposedly session timeout), ethersex goes wild and sends a new connect command every 400 ms until you reset hardware.

### Test code
* have a MQTT server running (with / without authentication doesn't make a difference)
* configure ethersex to connect to this MQTT server (no further scripts/traffic necessary) and wait until connected
* restart MQTT server
* wait some time (at max 60 seconds - supposedly the configured session timeout)

With authentication you can see a log like this:
>1538566270: mosquitto version 1.4.10 (build date Fri, 22 Dec 2017 08:19:25 +0000) starting
1538566270: Config loaded from /etc/mosquitto/mosquitto.conf.
1538566270: Opening ipv4 listen socket on port 1883.
1538566270: Opening ipv6 listen socket on port 1883.
1538566270: New connection from <etherxex-ip> on port 1883.
1538566270: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566270: New connection from <etherxex-ip> on port 1883.
1538566270: Socket error on client ethersex, disconnecting.
1538566271: New connection from <etherxex-ip> on port 1883.
1538566271: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566271: New connection from <etherxex-ip> on port 1883.
1538566271: Socket error on client ethersex, disconnecting.
1538566271: New connection from <etherxex-ip> on port 1883.
1538566271: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566360: Client <unknown> has exceeded timeout, disconnecting.
1538566360: Socket error on client <unknown>, disconnecting.
1538566360: New connection from <etherxex-ip> on port 1883.
1538566360: Client ethersex already connected, closing old connection.
1538566360: Client ethersex disconnected.
1538566360: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566360: New connection from <etherxex-ip> on port 1883.
1538566360: Client ethersex already connected, closing old connection.
1538566360: Client ethersex disconnected.
1538566360: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566361: Client <unknown> has exceeded timeout, disconnecting.
1538566361: Socket error on client <unknown>, disconnecting.
1538566361: New connection from <etherxex-ip> on port 1883.
1538566361: Client ethersex already connected, closing old connection.
1538566361: Client ethersex disconnected.
1538566361: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566361: New connection from <etherxex-ip> on port 1883.
1538566361: Client ethersex already connected, closing old connection.
1538566361: Client ethersex disconnected.
1538566361: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566361: New connection from <etherxex-ip> on port 1883.
1538566361: Client ethersex already connected, closing old connection.
1538566361: Client ethersex disconnected.
1538566361: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566361: New connection from <etherxex-ip> on port 1883.
1538566361: Client ethersex already connected, closing old connection.
1538566361: Client ethersex disconnected.
1538566361: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566362: New connection from <etherxex-ip> on port 1883.
1538566362: Client ethersex already connected, closing old connection.
1538566362: Client ethersex disconnected.
1538566362: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566362: New connection from <etherxex-ip> on port 1883.
1538566362: Client ethersex already connected, closing old connection.
1538566362: Client ethersex disconnected.
1538566362: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566362: New connection from <etherxex-ip> on port 1883.
1538566362: Client ethersex already connected, closing old connection.
1538566362: Client ethersex disconnected.
1538566362: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566362: New connection from <etherxex-ip> on port 1883.
1538566362: Client ethersex already connected, closing old connection.
1538566362: Client ethersex disconnected.
1538566362: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566362: New connection from <etherxex-ip> on port 1883.
1538566362: Client ethersex already connected, closing old connection.
1538566362: Client ethersex disconnected.
1538566362: New client connected from <etherxex-ip> as ethersex (c1, k60, u'ethersex').
1538566363: New connection from <etherxex-ip> on port 1883.
1538566363: Client ethersex already connected, closing old connection.
1538566363: Client ethersex disconnected.
[repeated indefinitely]

Without authentication, you would need Wireshark to see ethersex sending a "Connect Command" (and the server properly replying with "Connect Ack") every 400 ms

### Ethersex version
bb982f1e (not the latest, but no changes to MQTT/ethernet in last two changesets

Mosquitto 1.4.10 (Linux/Windows makes no difference)

### Hardware
ATMega32 on Pollin board

eku, 2018-10-03:
Please supply debug log from MQTT module (enable MQTTDEBUG).

stefan-muc, 2018-10-03:
MQTT running as it should.

Restart server:
>D: mqtt main poll
D: mqtt main poll
D: connection closed
D: initializing mqtt client
D: connection closed
D: initializing mqtt client
D: new connection
D: new connection
D: acked
D: received data:
D: connack received
D: mqtt main poll
D: mqtt main poll
D: connection closed
D: initializing mqtt client
D: connection closed
D: initializing mqtt client
D: new connection
D: new connection
D: acked
D: received data:
D: connack received
D: mqtt main poll
D: mqtt main poll
D: mqtt main poll
D: connection closed
D: initializing mqtt client
D: no uip_conn available.
D: connection closed
D: initializing mqtt client
D: new connection
D: acked
D: received data:
D: connack received
D: mqtt main poll

Wait some time... reply to a ping:
>D: mqtt main poll
D: ping request
D: mqtt main poll
D: mqtt main poll
D: acked
D: received data:
D: pingresp received
D: mqtt main poll

Wait some more time... go wild:
> D: mqtt main poll
D: connection closed
D: initializing mqtt client
D: no uip_conn available.
D: connection closed
D: initializing mqtt client
D: no uip_conn available.
D: connection closed
D: initializing mqtt client
D: connection closed
D: initializing mqtt client
D: new connection
D: new connection
D: connection closed
D: initializing mqtt client
D: no uip_conn available.
D: acked
D: received data:
D: unkown state
D: initializing mqtt client
D: no uip_conn available.
D: connection closed
D: initializing mqtt client
D: new connection
D: connection closed
D: initializing mqtt client
D: no uip_conn available.
D: acked
D: received data:
D: unkown state
D: initializing mqtt client
D: no uip_conn available.
D: connection closed
(repeat to infinity)

eku, 2019-03-14:
@stefan-muc: do not publish any new data while disconnected.

eku, 2019-03-18:
@stefan-muc,@mwue I put a simple publisher inside the mqtt poll callback, to make sure that the code only runs when the connection is active.
```
#define C6_TEST_PUBLISH_FORMAT   C6_MQTT_TOPIC "/test"
    static int counter = 0, value = 0;
    if (++counter == 100)
    { 
      counter = 0;
      value ^= 1;
      len = itoa_fixedpoint(value, 0, buf, sizeof(buf));
      C6DEBUG("poll %S=%s", PSTR(C6_TEST_PUBLISH_FORMAT), buf);
      mqtt_construct_publish_packet_P(PSTR(C6_TEST_PUBLISH_FORMAT), buf, len, false);
    } 
  }
```
A value is published every few seconds.

Then the server is stopped. Ethersex keeps trying to re-establish the connection. As soon as the server is running again, the connection is established and the value is published regularly.

I could not reproduce any abnormalities.

eku, 2019-03-31:
@stefan-muc,@mwue could you please try out my [branch](https://github.com/eku/ethersex/tree/fix_mqtt).

eku, 2019-06-23:
See also #433. Same wrong code in mqtt.c

stefan-muc, 2019-09-01:
I tested your code of #554 and sadly still was able to reproduce this issue.
Details: See IRC, ask to send it again if you didn't get it.

eku, 2019-12-05:
@stefan-muc As discussed in IRC, I cannot reproduce the behavior with a Linux MQTT server. The behavior described here was observed with a Windows MQTT server.

In the provided network recording the following picture can be observed after restarting the MQTT server. The MQTT client reconnects. After the TCP connection has been established, an MQTT connect is sent, which the server also acknowledges and then closes the TCP connection.

I can't see an error on the client side. Please activate debug logging on the Windows MQTT server and attach it here. Maybe there is a reason for terminating the connection.

eku, 2020-04-27:
No response for five months. The changes of PR#554 are now merged. If there are still problems, please reopen this issue.

eku, 2020-10-29:
@stefan-muc I just managed that E6 can't establish a new connection to the MQTT broker. In Wireshark you see a SYN packet plus retransmits of it and finally a SYS-RST packet without the broker ever responding. It is not yet clear to me what the reason is.

eku, 2026-08-20:
Fixed with 4000b9bb.

closed 2026-08-20 by eku